mbeddr/mbeddr.core #2281: Doc language: exceptions after migration to 2022.2

alexanderpann, 2023-06-12:
Some exceptions are thrown by this language that appear in the log file.
@HeikoBecker can you edit the description and add the exception from your log file?

HeikoBecker, 2023-06-13:
It seems that this issue has been solved with the latest fixes from MPS-extensions. Closing this for now, will reopen if the errors appear again.

alexanderpann, 2023-06-13:
I am not so convinced, but I haven't checked. One of the error messages was also fixed in MPS 2022.2.1:
`You must not register toolwindow programmatically so early. Rework code or use ToolWindowManager.invokeLater` ([MPS-35465](https://youtrack.jetbrains.com/issue/MPS-35465))

HeikoBecker, 2023-06-14:
With some fiddling around in files I could get this to appear in the log:

```
2023-06-14 10:26:56,607 [2607584]   WARN - #c.i.o.a.Presentation - Shall not be called on a template presentation
java.lang.Throwable: Shall not be called on a template presentation
        at com.intellij.openapi.actionSystem.Presentation.assertNotTemplatePresentation(Presentation.java:450)
        at com.intellij.openapi.actionSystem.Presentation.setEnabled(Presentation.java:423)
        at jetbrains.mps.workbench.action.BaseAction.disable(BaseAction.java:256)
        at jetbrains.mps.workbench.action.BaseAction.setEnabledState(BaseAction.java:271)
        at jetbrains.mps.lang.generator.pluginSolution.plugin.NewCreateRootRule_Action.doUpdate(NewCreateRootRule_Action.java:66)
        at jetbrains.mps.workbench.action.BaseAction.lambda$update$0(BaseAction.java:176)
        at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:80)
        at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
        at jetbrains.mps.workbench.action.BaseAction.update(BaseAction.java:165)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu.collectActionsAsIntentions(OriginalIntentionMenu.java:203)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu.collectActionsAsIntentions(OriginalIntentionMenu.java:195)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu.collectActionsAsIntentions(OriginalIntentionMenu.java:195)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu.getIntentionsGroup(OriginalIntentionMenu.java:163)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu$7.compute(OriginalIntentionMenu.java:216)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu$7.compute(OriginalIntentionMenu.java:212)
        at jetbrains.mps.util.ComputeRunnable.compute(ComputeRunnable.java:32)
        at jetbrains.mps.util.AbstractComputeRunnable.run(AbstractComputeRunnable.java:27)
        at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:80)
        at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
        at jetbrains.mps.smodel.ModelAccessHelper.runReadAction(ModelAccessHelper.java:44)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu.showIntentionsMenu(OriginalIntentionMenu.java:212)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu.checkAndShowMenu(OriginalIntentionMenu.java:100)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu$2$1.run(OriginalIntentionMenu.java:84)
        at jetbrains.mps.smodel.ActionDispatcher.dispatch(ActionDispatcher.java:98)
        at jetbrains.mps.smodel.ActionDispatcher.lambda$wrap$0(ActionDispatcher.java:135)
        at jetbrains.mps.smodel.LockRunnable.run(LockRunnable.java:60)
        at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:915)
        at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:84)
        at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
        at com.mbeddr.mpsutil.intentions.runtime.OriginalIntentionMenu$2.actionPerformed(OriginalIntentionMenu.java:81)
        at java.desktop/javax.swing.JComponent$ActionStandin.actionPerformed(JComponent.java:3493)
        at java.desktop/javax.swing.SwingUtilities.notifyAction(SwingUtilities.java:1810)
        at java.desktop/javax.swing.JComponent.processKeyBinding(JComponent.java:2950)
        at java.desktop/javax.swing.JComponent.processKeyBindings(JComponent.java:3012)
        at java.desktop/javax.swing.JComponent.processKeyEvent(JComponent.java:2912)
        at java.desktop/java.awt.Component.processEvent(Component.java:6425)
        at java.desktop/java.awt.Container.processEvent(Container.java:2266)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:5022)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4854)
        at java.desktop/java.awt.KeyboardFocusManager.redispatchEvent(KeyboardFocusManager.java:1964)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchKeyEvent(DefaultKeyboardFocusManager.java:886)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.preDispatchKeyEvent(DefaultKeyboardFocusManager.java:1166)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.typeAheadAssertions(DefaultKeyboardFocusManager.java:1023)
        at java.desktop/java.awt.DefaultKeyboardFocusManager.dispatchEvent(DefaultKeyboardFocusManager.java:849)
        at com.intellij.ide.IdeKeyboardFocusManager.lambda$dispatchEvent$0(IdeKeyboardFocusManager.java:48)
        at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
        at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:604)
        at com.intellij.ide.IdeKeyboardFocusManager.dispatchEvent(IdeKeyboardFocusManager.java:48)
        at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4903)
        at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
        at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2802)
        at java.desktop/java.awt.Component.dispatchEvent(Component.java:4854)
[... cut ...]
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
        at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
        at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
        at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
```

Will check whether this is genuine, re-opening.

HeikoBecker, 2023-06-14:
The bug I reported above is coming from mps-extensions. Will move it there. 

PR #2282 addresses some other bugs I found while investigating.

HeikoBecker, 2023-06-20:
Another bug found while working with mbeddr.core:

```User's action execute method failed. Action: Open Solution Browser. Class: com.mbeddr.core.base.pluginSolution.plugin.openSolutionTree2_Action

java.lang.NullPointerException: Cannot invoke "javax.swing.JTree.getRowCount()" because "this.tree" is null
	at com.mbeddr.core.base.pluginSolution.plugin.TreeExpansionHelper.captureExpansionState(TreeExpansionHelper.java:19)
	at com.mbeddr.core.base.pluginSolution.plugin.GenericTreeTool_Tool.focusOn(GenericTreeTool_Tool.java:78)
	at com.mbeddr.core.base.pluginSolution.plugin.GenericTreeTool_Tool.changeRootTo(GenericTreeTool_Tool.java:71)
	at com.mbeddr.core.base.pluginSolution.plugin.GenericTreeTool_Tool.setRoot(GenericTreeTool_Tool.java:60)
	at com.mbeddr.core.base.pluginSolution.plugin.openSolutionTree2_Action.doExecute(openSolutionTree2_Action.java:69)
	at jetbrains.mps.workbench.action.BaseAction.lambda$actionPerformed$2(BaseAction.java:208)
	at jetbrains.mps.workbench.action.ActionAccess$CommandProjectAccess$1.run(ActionAccess.java:76)
	at jetbrains.mps.smodel.WorkbenchModelAccess$UndoContextSetup.run(WorkbenchModelAccess.java:367)
	at jetbrains.mps.smodel.ActionDispatcher.dispatch(ActionDispatcher.java:98)
	at jetbrains.mps.smodel.ActionDispatcher.lambda$wrap$0(ActionDispatcher.java:135)
	at jetbrains.mps.smodel.ActionDispatcher.dispatch(ActionDispatcher.java:98)
	at jetbrains.mps.smodel.ActionDispatcher.lambda$wrap$0(ActionDispatcher.java:135)
	at jetbrains.mps.smodel.LockRunnable.run(LockRunnable.java:60)
	at com.intellij.openapi.application.impl.ApplicationImpl.runWriteAction(ApplicationImpl.java:1023)
	at jetbrains.mps.smodel.TryRunPlatformWriteHelper.runWrite(TryRunPlatformWriteHelper.java:103)
	at jetbrains.mps.smodel.TryRunPlatformWriteHelper.lambda$withPlatformWrite$1(TryRunPlatformWriteHelper.java:96)
	at com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:219)
	at com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:174)
	at com.intellij.openapi.command.impl.CoreCommandProcessor.executeCommand(CoreCommandProcessor.java:164)
	at jetbrains.mps.smodel.WorkbenchModelAccess.executeCommand(WorkbenchModelAccess.java:302)
	at jetbrains.mps.smodel.ProjectModelAccess2.executeCommand(ProjectModelAccess2.java:35)
	at jetbrains.mps.workbench.action.ActionAccess$CommandProjectAccess.runWithAccess(ActionAccess.java:73)
	at jetbrains.mps.workbench.action.BaseAction.actionPerformed(BaseAction.java:200)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.doPerformActionOrShowPopup(ActionUtil.java:315)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.lambda$performActionDumbAwareWithCallbacks$4(ActionUtil.java:294)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performDumbAwareWithCallbacks(ActionUtil.java:337)
	at com.intellij.openapi.actionSystem.ex.ActionUtil.performActionDumbAwareWithCallbacks(ActionUtil.java:294)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.lambda$performAction$4(ActionMenuItem.java:293)
	at com.intellij.openapi.wm.impl.FocusManagerImpl.runOnOwnContext(FocusManagerImpl.java:224)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.performAction(ActionMenuItem.java:286)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.lambda$new$0(ActionMenuItem.java:67)
	at java.desktop/javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:1972)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.lambda$fireActionPerformed$3(ActionMenuItem.java:109)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
	at com.intellij.openapi.application.TransactionGuardImpl.performUserActivity(TransactionGuardImpl.java:94)
	at com.intellij.openapi.actionSystem.impl.ActionMenuItem.fireActionPerformed(ActionMenuItem.java:109)
	at com.intellij.ui.plaf.beg.BegMenuItemUI.doClick(BegMenuItemUI.java:546)
	at com.intellij.ui.plaf.beg.BegMenuItemUI$MyMenuDragMouseHandler.menuDragMouseReleased(BegMenuItemUI.java:610)
	at java.desktop/javax.swing.JMenuItem.fireMenuDragMouseReleased(JMenuItem.java:587)
	at java.desktop/javax.swing.JMenuItem.processMenuDragMouseEvent(JMenuItem.java:484)
	at java.desktop/javax.swing.JMenuItem.processMouseEvent(JMenuItem.java:429)
	at java.desktop/javax.swing.MenuSelectionManager.processMouseEvent(MenuSelectionManager.java:347)
	at java.desktop/javax.swing.plaf.basic.BasicPopupMenuUI$MouseGrabber.eventDispatched(BasicPopupMenuUI.java:905)
	at java.desktop/java.awt.Toolkit$SelectiveAWTEventListener.eventDispatched(Toolkit.java:2202)
	at java.desktop/java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2094)
	at java.desktop/java.awt.Toolkit$ToolkitEventMulticaster.eventDispatched(Toolkit.java:2093)
	at java.desktop/java.awt.Toolkit.notifyAWTEventListeners(Toolkit.java:2050)
	at java.desktop/java.awt.Component.dispatchEventImpl(Component.java:4928)
	at java.desktop/java.awt.Container.dispatchEventImpl(Container.java:2324)
	at java.desktop/java.awt.Window.dispatchEventImpl(Window.java:2802)
	at java.desktop/java.awt.Component.dispatchEvent(Component.java:4854)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:781)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
[... cut ...]
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)

```

PR #2282 (2023-07-06): Fix bugs left from migration to 2022.2
This PR addresses some internal changes that were required to fix bugs originating from the 2022.2 migration. 

Closes #2281
closed 2023-07-06 by HeikoBecker