mbeddr/mbeddr.core #2287: com.mbeddr.mpsutil.refactoring not available in JetBrains plugin repository

eugenschindler, 2023-07-15:
I can't find the com.mbeddr.mpsutil.refactoring language (needed for me mainly for the ProjectionModeSwitcher) in the JetBrains plugin repository. It could be made available, but it may also be considered to factor out the ProjectionModeSwitcher into a separate util (since from the outside it doesn't look like it logically belongs to "refactoring" utils) and then publish it on the JetBrains repository.

eugenschindler, 2023-07-15:
@alexanderpann I would be very interested to hear your thoughts about this issue/idea.

alexanderpann, 2023-07-16:
Not a single plugin from the mbeddr platform is published in the JetBrains plugin repository. The only exception is plugins that moved to MPS Extensions such as `com.​mbeddr.​mpsutil.​projectview` which kept the old package name. Only publishing the refactoring plugin would be kinda random, all plugins from the `com.mbeddr.platform` [buildscript](http://127.0.0.1:63320/node?ref=r%3A742f344d-4dc4-4862-992c-4bc94b094870%28com.mbeddr.mpsutil.dev.build%29%2F4160126263863194133) should be published if we do it properly but that would be a lot of effort.

Extracting it into a plugin makes sense. It is a breaking change where you would have to deprecate the old concepts and keep them working for a while so that you don't break any projects that depend on it.