mbeddr/mbeddr.core #2288: Many mbeddr.core languages don't load properly when pulled in via global libraries

eugenschindler, 2023-07-15:
Many of the mbeddr.core (not the platform) languages give problems with being properly class-loaded when I just put them into a subdirectory on my system and then point to them via global libraries (even after rebooting MPS).
If I copy all those languages in the plugins of the MPS isntallation itself, then they all load fine.
Maybe some custom class-loading goes wrong when (at least part of) the mbeddr.core languages are loaded via global libraries (and I guess also project libraries?).

alexanderpann, 2023-07-16:
The plugin `com.mbeddr.spawner` (debugger plugin) which is an IntelliJ platform probably needs to be loaded from the `plugins` folder to work properly. Some of the other languages need this plugin loaded so that they are loaded themselves.

eugenschindler, 2024-08-26:
Thanks for the answer. What can I best do to make these issues go away? And is it possible to fix it also for the case when plugins are not copied into the plugins folder of MPS?

alexanderpann, 2024-08-29:
Except copying the spawner plugin to the plugins folder or voting to remove the debugger, not much. There is only a version 1.0 and 1.1 of this plugin, so you can also use the same version of it. When you check out the mbeddr repository itself, you should be able to install it with `./gradlew install` or `gradlew.bat install` where you might need to set the parameter `-PMPS_PATHS_SELECTOR` to the MPS home path.
> And is it possible to fix it also for the case when plugins are not copied into the plugins folder of MPS?

What do you mean?

alexanderpann, 2024-11-25:
The debugger was removed in https://github.com/mbeddr/mbeddr.core/pull/2509. The ticket is now obsolete.

closed 2024-11-25 by alexanderpann