mbeddr/mbeddr.core #2404: NullPointerException caused by  at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler

arimer, 2024-04-08:
We are using the mbeddr.platform in version 2022.2.24492.bc2ece9 and are seeing the following exception on MPS startup:

```
java.lang.NullPointerException: Cannot invoke "com.intellij.openapi.wm.ToolWindow.isVisible()" because the return value of "jetbrains.mps.ide.tools.BaseTool.getToolWindow()" is null
    at jetbrains.mps.ide.tools.BaseTool.toolIsOpened(BaseTool.java:101)
    at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler.lambda$showDoc$0(DocumentationHandler.java:63)
    at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:80)
    at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
    at com.mbeddr.doc.aspect.ui.plugin.DocumentationHandler.showDoc(DocumentationHandler.java:38)
    at com.mbeddr.doc.aspect.ui.plugin.DocumentationProjectPlugin_ProjectPluginPart$1.selectionChangedTo(DocumentationProjectPlugin_ProjectPluginPart.java:25)
    at jetbrains.mps.nodeEditor.selection.SingularSelectionListenerAdapter.selectionChanged(SingularSelectionListenerAdapter.java:33)
    at jetbrains.mps.nodeEditor.selection.SelectionManagerImpl.doChangeSelection(SelectionManagerImpl.java:219)
    at jetbrains.mps.nodeEditor.selection.SelectionManagerImpl.setSelectionInfoStack(SelectionManagerImpl.java:176)
    at jetbrains.mps.nodeEditor.Memento.restore(Memento.java:185)
    at jetbrains.mps.nodeEditor.EditorContext.lambda$restoreEditorComponentState$0(EditorContext.java:182)
    at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:80)
    at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
    at jetbrains.mps.nodeEditor.EditorContext.restoreEditorComponentState(EditorContext.java:180)
    at jetbrains.mps.nodeEditor.updater.UpdaterImpl.doUpdate(UpdaterImpl.java:115)
    at jetbrains.mps.nodeEditor.updater.UpdaterImpl.update(UpdaterImpl.java:90)
    at jetbrains.mps.nodeEditor.EditorComponent.rebuildEditorContent(EditorComponent.java:2015)
    at jetbrains.mps.nodeEditor.NodeEditorComponent.rebuildEditorContent(NodeEditorComponent.java:126)
    at jetbrains.mps.nodeEditor.EditorComponent$1.lambda$onLoaded$0(EditorComponent.java:300)
    at jetbrains.mps.smodel.ActionDispatcher.dispatch(ActionDispatcher.java:98)
    at jetbrains.mps.smodel.ActionDispatcher.lambda$wrap$0(ActionDispatcher.java:135)
    at jetbrains.mps.smodel.LockRunnable.run(LockRunnable.java:60)
    at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:915)
    at jetbrains.mps.smodel.WorkbenchModelAccess.tryRead(WorkbenchModelAccess.java:165)
    at jetbrains.mps.smodel.WorkbenchModelAccess.lambda$runReadInEDT$0(WorkbenchModelAccess.java:122)
    at jetbrains.mps.smodel.EDTExecutorInternal.tryToRunTopTask(EDTExecutorInternal.java:296)
    at jetbrains.mps.smodel.EDTExecutorInternal.flushNTasks(EDTExecutorInternal.java:251)
    at jetbrains.mps.smodel.EDTExecutorInternal.doFlush(EDTExecutorInternal.java:233)
    at jetbrains.mps.smodel.EDTExecutorInternal.flushTasksQueue(EDTExecutorInternal.java:213)
    at jetbrains.mps.smodel.EDTExecutorInternal.lambda$guaranteeWriteSafetyViaHack$2(EDTExecutorInternal.java:200)
    at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
    at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
    at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
    at jetbrains.mps.smodel.EDTExecutorInternal.lambda$guaranteeWriteSafetyViaHack$3(EDTExecutorInternal.java:205)
    at jetbrains.mps.smodel.EDTExecutorInternal.lambda$forceScheduleFlushEDT$1(EDTExecutorInternal.java:168)
    at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:194)
    at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
    at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:513)
    at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:75)
    at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:118)
    at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
    at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
    at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:779)
    at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
    at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
    at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
    at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
    at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:749)
    at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:918)
    at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:766)
    at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:450)
    at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:791)
    at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:449)
    at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:113)
    at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:624)
    at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:447)
    at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
    at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:493)
    at com.intellij.ide.IdeEventQueue.pumpEventsForHierarchy(IdeEventQueue.java:965)
    at com.intellij.openapi.progress.util.ProgressWindow.lambda$startBlocking$4(ProgressWindow.java:211)
    at com.intellij.openapi.application.impl.ApplicationImpl.runUnlockingIntendedWrite(ApplicationImpl.java:907)
    at com.intellij.openapi.progress.util.ProgressWindow.startBlocking(ProgressWindow.java:206)
    at com.intellij.openapi.progress.impl.ProgressRunner.lambda$execFromEDT$6(ProgressRunner.java:329)
    at java.base/java.util.concurrent.CompletableFuture.uniAcceptNow(CompletableFuture.java:757)
    at java.base/java.util.concurrent.CompletableFuture.uniAcceptStage(CompletableFuture.java:735)
    at java.base/java.util.concurrent.CompletableFuture.thenAccept(CompletableFuture.java:2182)
[... 15 lines cut ...]
    at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
    at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
    at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:779)
    at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
    at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
    at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
    at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
    at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:749)
    at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:918)
    at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:766)
    at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:450)
    at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:791)
    at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:449)
    at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
    at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:624)
    at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:447)
    at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
    at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:493)
    at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
    at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
    at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
    at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
    at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
    at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)
```

alexanderpann, 2024-04-08:
Did you try the latest version and check if the fix in #2401 helps?

arimer, 2024-04-08:
Will do.

mgronover, 2024-04-16:
With the #2401 we exchange the error message:

```
Exception on a tool init: Tool com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool

java.lang.IllegalStateException: You must not register toolwindow programmatically so early. Rework code or use ToolWindowManager.invokeLater
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.getToolWindowPaneIfInitialized(ToolWindowManagerImpl.kt:500)
	at com.intellij.openapi.wm.impl.ToolWindowManagerImpl.registerToolWindow(ToolWindowManagerImpl.kt:930)
	at com.intellij.openapi.wm.ToolWindowManager.registerToolWindow(ToolWindowManager.kt:96)
	at jetbrains.mps.ide.tools.BaseTool.register(BaseTool.java:268)
	at jetbrains.mps.ide.tools.BaseTool.getToolWindow(BaseTool.java:198)
	at jetbrains.mps.ide.tools.BaseTool.toolIsOpened(BaseTool.java:101)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.initializeTabsForSelection(TabbedDocViewer_Tool.java:83)
	at com.mbeddr.doc.aspect.ui.plugin.TabbedDocViewer_Tool.init(TabbedDocViewer_Tool.java:68)
	at jetbrains.mps.plugins.projectplugins.BaseProjectPlugin.registerPrefsAndTools(BaseProjectPlugin.java:130)
	at jetbrains.mps.plugins.projectplugins.BaseProjectPlugin.init(BaseProjectPlugin.java:97)
	at jetbrains.mps.plugins.projectplugins.ProjectPluginManager.createPlugin(ProjectPluginManager.java:159)
	at jetbrains.mps.plugins.projectplugins.ProjectPluginManager.createPlugin(ProjectPluginManager.java:54)
	at jetbrains.mps.plugins.BasePluginManager.createPluginChecked(BasePluginManager.java:153)
	at jetbrains.mps.plugins.BasePluginManager.createPlugins(BasePluginManager.java:143)
	at jetbrains.mps.plugins.BasePluginManager.loadPlugins(BasePluginManager.java:73)
	at jetbrains.mps.plugins.PluginLoaderRegistry.loadContributors(PluginLoaderRegistry.java:224)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.addLoaders(PluginLoaderRegistry.java:570)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.lambda$update$2(PluginLoaderRegistry.java:531)
	at jetbrains.mps.classloading.ClassLoaderManager.lambda$runTransaction$2(ClassLoaderManager.java:314)
	at jetbrains.mps.classloading.ClassLoaderManager.runTransaction(ClassLoaderManager.java:322)
	at jetbrains.mps.classloading.ClassLoaderManager.runTransaction(ClassLoaderManager.java:313)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.lambda$update$3(PluginLoaderRegistry.java:478)
	at jetbrains.mps.smodel.ActionDispatcher.dispatch(ActionDispatcher.java:98)
	at jetbrains.mps.smodel.ActionDispatcher.lambda$wrap$0(ActionDispatcher.java:135)
	at jetbrains.mps.smodel.LockRunnable.run(LockRunnable.java:60)
	at com.intellij.openapi.application.impl.ApplicationImpl.runReadAction(ApplicationImpl.java:915)
	at jetbrains.mps.smodel.WorkbenchModelAccess.runReadAction(WorkbenchModelAccess.java:84)
	at jetbrains.mps.smodel.ModelAccessBase.runReadAction(ModelAccessBase.java:56)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.update(PluginLoaderRegistry.java:477)
	at jetbrains.mps.plugins.PluginLoaderRegistry$UpdatingTask.lambda$run$1(PluginLoaderRegistry.java:463)
	at com.intellij.openapi.application.TransactionGuardImpl.runWithWritingAllowed(TransactionGuardImpl.java:209)
	at com.intellij.openapi.application.TransactionGuardImpl.access$100(TransactionGuardImpl.java:21)
	at com.intellij.openapi.application.TransactionGuardImpl$1.run(TransactionGuardImpl.java:191)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
	at com.intellij.openapi.application.impl.ApplicationImpl$3.run(ApplicationImpl.java:513)
	at com.intellij.openapi.application.impl.LaterInvocator$1.run(LaterInvocator.java:95)
	at com.intellij.openapi.application.impl.FlushQueue.doRun(FlushQueue.java:75)
	at com.intellij.openapi.application.impl.FlushQueue.runNextEvent(FlushQueue.java:118)
	at com.intellij.openapi.application.impl.FlushQueue.flushNow(FlushQueue.java:42)
	at java.desktop/java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:318)
	at java.desktop/java.awt.EventQueue.dispatchEventImpl(EventQueue.java:779)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:730)
	at java.desktop/java.awt.EventQueue$4.run(EventQueue.java:724)
	at java.base/java.security.AccessController.doPrivileged(AccessController.java:399)
	at java.base/java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:86)
	at java.desktop/java.awt.EventQueue.dispatchEvent(EventQueue.java:749)
	at com.intellij.ide.IdeEventQueue.defaultDispatchEvent(IdeEventQueue.java:918)
	at com.intellij.ide.IdeEventQueue._dispatchEvent(IdeEventQueue.java:766)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$6(IdeEventQueue.java:450)
	at com.intellij.openapi.progress.impl.CoreProgressManager.computePrioritized(CoreProgressManager.java:791)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$7(IdeEventQueue.java:449)
	at com.intellij.openapi.application.TransactionGuardImpl.performActivity(TransactionGuardImpl.java:105)
	at com.intellij.ide.IdeEventQueue.performActivity(IdeEventQueue.java:624)
	at com.intellij.ide.IdeEventQueue.lambda$dispatchEvent$8(IdeEventQueue.java:447)
	at com.intellij.openapi.application.impl.ApplicationImpl.runIntendedWriteActionOnCurrentThread(ApplicationImpl.java:881)
	at com.intellij.ide.IdeEventQueue.dispatchEvent(IdeEventQueue.java:493)
	at java.desktop/java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:207)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:128)
	at java.desktop/java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:117)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:113)
	at java.desktop/java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:105)
	at java.desktop/java.awt.EventDispatchThread.run(EventDispatchThread.java:92)

```