mbeddr/mbeddr.core #2439: Replace -XX:MaxPermSize with -XX:MaxMetaspaceSize in build scripts (2022.3)

alexanderpann, 2024-05-28:
The option was removed in the OpenJDK 17 and means that people can't build mbeddr on Linux.

alexanderpann, 2024-05-28:
We actually start the build with a JDK 11 on TeamCity:
`/usr/lib/jvm/11_0_14_1-b1751.46/bin/java --add-opens=java.base/java.util=ALL-UNNAMED --add-opens=java.base/java.lang=ALL-UNNAMED --add-opens=java.base/java.lang.invoke=ALL-UNNAMED --add-opens=java.prefs/java.util.prefs=ALL-UNNAMED --add-opens=java.base/java.nio.charset=ALL-UNNAMED --add-opens=java.base/java.net=ALL-UNNAMED --add-opens=java.base/java.util.concurrent.atomic=ALL-UNNAMED -XX:MaxMetaspaceSize=384m -XX:+HeapDumpOnOutOfMemoryError -Xms256m -Xmx512m -Dfile.encoding=UTF-8 -Djava.io.tmpdir=/build/temp/buildTmp -Duser.country -Duser.language=en -Duser.variant -cp /root/.gradle/wrapper/dists/gradle-8.0.2-bin/`

Then we run the build with the downloaded JBR which is JBR 17. This option is not added in mbeddr itself. Do you maybe have a gradle.properties file (maybe in ~./gradle) that sets this option? As you can see, on the CI is uses the `MaxMetaspaceSize` option.

ARaack, 2024-05-29:
Yes your're right, in my local ~./gradle/gradle.properties was a setting with that option. 
Will test it again.

ARaack, 2024-05-29:
Update, alright all works like expected. Sorry for the inconvenience. 
Issue can be closed.