mbeddr/mbeddr.core #2535: MPS 2023.2: "Save settings failed" error related to custom project view

sergej-koscejev, 2024-09-24:
In MPS 2023.2 when opening a project with custom project view, occasionally an error balloon appears, with the title "Unable to save settings" and message "Failed to save settings. Please restart JetBrains MPS". The following exception appears in the logs:

```
2024-09-24 13:16:47,407 [2076087]   WARN - #c.i.o.c.i.s.StoreUtil - Save settings failed
java.lang.Exception: Cannot get ProjectView component state
	at com.intellij.configurationStore.ComponentStoreImpl.commitComponents$suspendImpl(ComponentStoreImpl.kt:249)
	at com.intellij.configurationStore.ComponentStoreImpl$commitComponents$1.invokeSuspend(ComponentStoreImpl.kt)
	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
	at kotlinx.coroutines.scheduling.CoroutineScheduler.runSafely(CoroutineScheduler.kt:584)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.executeTask(CoroutineScheduler.kt:793)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.runWorker(CoroutineScheduler.kt:697)
	at kotlinx.coroutines.scheduling.CoroutineScheduler$Worker.run(CoroutineScheduler.kt:684)
Caused by: jetbrains.mps.classloading.ModuleClassLoader$ModuleClassLoaderIsDisposedException: ClassLoader of the module 'com.mbeddr.mpsutil.projectview.runtime [solution]' is disposed and not operable!
	at jetbrains.mps.classloading.ModuleClassLoader.checkNotDisposed(ModuleClassLoader.java:87)
	at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:118)
	at jetbrains.mps.classloading.ModuleClassLoader.loadClass(ModuleClassLoader.java:114)
	at java.base/java.lang.ClassLoader.loadClass(ClassLoader.java:525)
	at c.m.m.projectview.runtime//com.mbeddr.mpsutil.projectview.runtime.tree.CustomProjectView.writeExternal(CustomProjectView.java:570)
	at com.intellij.ide.projectView.impl.ProjectViewImpl.writePaneState(ProjectViewImpl.java:1516)
	at com.intellij.ide.projectView.impl.ProjectViewImpl.getState(ProjectViewImpl.java:1506)
	at jetbrains.mps.ide.projectView.ProjectViewImpl.getState(ProjectViewImpl.java:68)
	at jetbrains.mps.ide.projectView.ProjectViewImpl.getState(ProjectViewImpl.java:31)
	at com.intellij.configurationStore.ComponentStoreImpl$commitComponent$5.invoke(ComponentStoreImpl.kt:364)
	at com.intellij.openapi.application.rw.InternalReadAction.insideReadAction(InternalReadAction.kt:105)
	at com.intellij.openapi.application.rw.InternalReadAction.access$insideReadAction(InternalReadAction.kt:14)
	at com.intellij.openapi.application.rw.InternalReadAction$tryReadCancellable$2.invoke(InternalReadAction.kt:95)
	at com.intellij.openapi.application.rw.InternalReadAction$tryReadCancellable$2.invoke(InternalReadAction.kt:94)
	at com.intellij.openapi.application.rw.CancellableReadActionKt$cancellableReadActionInternal$1.invoke$lambda$1$lambda$0(cancellableReadAction.kt:41)
	at com.intellij.openapi.application.impl.ApplicationImpl.tryRunReadAction(ApplicationImpl.java:1133)
	at com.intellij.openapi.application.rw.CancellableReadActionKt$cancellableReadActionInternal$1.invoke$lambda$1(cancellableReadAction.kt:39)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtilService.runActionAndCancelBeforeWrite(ProgressIndicatorUtilService.java:63)
	at com.intellij.openapi.progress.util.ProgressIndicatorUtils.runActionAndCancelBeforeWrite(ProgressIndicatorUtils.java:133)
	at com.intellij.openapi.application.rw.CancellableReadActionKt$cancellableReadActionInternal$1.invoke(cancellableReadAction.kt:37)
	at com.intellij.openapi.progress.CoroutinesKt.blockingContext(coroutines.kt:248)
	at com.intellij.openapi.application.rw.CancellableReadActionKt.cancellableReadActionInternal(cancellableReadAction.kt:34)
	at com.intellij.openapi.application.rw.InternalReadAction.tryReadCancellable(InternalReadAction.kt:94)
	at com.intellij.openapi.application.rw.InternalReadAction.tryReadAction(InternalReadAction.kt:76)
	at com.intellij.openapi.application.rw.InternalReadAction.readLoop(InternalReadAction.kt:63)
	at com.intellij.openapi.application.rw.InternalReadAction.access$readLoop(InternalReadAction.kt:14)
	at com.intellij.openapi.application.rw.InternalReadAction$runReadAction$4.invokeSuspend(InternalReadAction.kt:42)
	at com.intellij.openapi.application.rw.InternalReadAction$runReadAction$4.invoke(InternalReadAction.kt)
	at com.intellij.openapi.application.rw.InternalReadAction$runReadAction$4.invoke(InternalReadAction.kt)
	at kotlinx.coroutines.intrinsics.UndispatchedKt.startUndispatchedOrReturn(Undispatched.kt:78)
	at kotlinx.coroutines.BuildersKt__Builders_commonKt.withContext(Builders.common.kt:167)
	at kotlinx.coroutines.BuildersKt.withContext(Unknown Source)
	at com.intellij.openapi.application.rw.InternalReadAction.runReadAction(InternalReadAction.kt:38)
	at com.intellij.openapi.application.rw.PlatformReadWriteActionSupport.executeReadAction(PlatformReadWriteActionSupport.kt:38)
	at com.intellij.openapi.application.ReadWriteActionSupport.executeReadAction$default(ReadWriteActionSupport.kt:15)
	at com.intellij.openapi.application.CoroutinesKt.constrainedReadAction(coroutines.kt:58)
	at com.intellij.openapi.application.CoroutinesKt.readAction(coroutines.kt:25)
	at com.intellij.configurationStore.ComponentStoreImpl.commitComponent(ComponentStoreImpl.kt:363)
	at com.intellij.configurationStore.ComponentStoreImpl.commitComponents$suspendImpl(ComponentStoreImpl.kt:245)
	... 7 more
```

alexanderpann, 2024-10-18:
Btw, wrong project but fixed in https://github.com/JetBrains/MPS-extensions/pull/986

closed 2024-10-18 by alexanderpann

sergej-koscejev, 2024-10-21:
Thanks. It's difficult for me to tell which module is in which project as long as MPS-extensions is merged with mbeddr platform during the build.