mbeddr/mbeddr.core #2659: "Generation required for TraceExplorer" message on any build

enikao, 2024-12-17:
When using mbeddr.core, any time I build something I get the message:

```
Module com.mbeddr.mpsutil.traceExplorer from dependencies requires compilation
```

Tried on iets3.opensource.

MPS 2023.2.2
com.mbeddr.platform.build=2023.2.25061.ecd4e3e

enikao, 2024-12-18:
Haven't looked into details yet. I didn't see any immediate issues.

The weird thing is I use the packaged version of mbeddr.platform.

alexanderpann, 2025-03-12:
This warning message is related to https://youtrack.jetbrains.com/issue/MPS-37220:
The relevant commit from MPS:
```
If a module from dependencies is reported DIRTY, it is left in components/cycles to build.
As we collect JM dependencies for the initial set of modules only, there are no chances to get
the module from dependencies compiled.
```

The only dependency the language has is on the interpreter.