mbeddr/mbeddr.core #2670: Deployed artifact contains unwanted stuff in icons folder

kbirken, 2024-12-20:
When mbeddr.platform is loaded into the `build/dependencies` folder by gradle, there is a subfolder `com.mbeddr.platform/mbeddr.mpsutil/icons` which contains stuff that should not be part of the delivery. Here are the examples which triggered me, but there might be more:

```
./mbeddr.mpsutil/icons/test.com.mbeddr.mpsutil.logicalChild
./mbeddr.mpsutil/icons/test.com.mbeddr.mpsutil.logicalChild/models/test.com.mbeddr.mpsutil.logicalChild.test@tests.mps
./mbeddr.mpsutil/icons/test.com.mbeddr.mpsutil.logicalChild/test.com.mbeddr.mpsutil.logicalChild.msd
./mbeddr.mpsutil/icons/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox
./mbeddr.mpsutil/icons/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox/models/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox.testbuffer.mps
./mbeddr.mpsutil/icons/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox/models/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox.example.mps
./mbeddr.mpsutil/icons/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox/com.mbeddr.mpsutil.logicalChild.sandbox.sandbox.msd
```

These shouldn't be part of the deployed artefacts at all. In this particular case, it leads to warnings in customer projects, e.g.

```
2024-12-20 15:57:20,326 [1688805]   WARN - jetbrains.mps.classloading.ModulesWatcher - com.mbeddr.mpsutil.logicalChild.sandbox.sandbox was marked invalid for class loading: The dependency in com.mbeddr.mpsutil.logicalChild.sandbox.sandbox [solution] cannot be resolved {Default} 286e2375-00e3-4042-b083-84873dd623be(com.mbeddr.mpsutil.logicalChild.sandbox)
2024-12-20 15:57:20,326 [1688805]   WARN - jetbrains.mps.classloading.ModulesWatcher - test.com.mbeddr.mpsutil.logicalChild was marked invalid for class loading: The dependency in test.com.mbeddr.mpsutil.logicalChild [solution] 
```

I don't know what has to be changed to avoid this, otherwise I would provide a PR for it.

alexanderpann, 2024-12-21:
It think it is caused by [this line](http://127.0.0.1:63320/node?ref=r%3A742f344d-4dc4-4862-992c-4bc94b094870%28com.mbeddr.mpsutil.dev.build%29%2F7984302635364029381) in the build script:  folder icons... files from $mpsutil/solutions

kbirken, 2025-01-21:
PR has been merged, the icons-folder now only contains icons files and no more complete solutions etc. The warnings mentioned in the description also are gone after the change.

closed 2025-01-21 by kbirken