mbeddr/mbeddr.core #2844: PlantUML visualisation not workint with MPS 2023.2

arimer, 2025-04-01:
This error appears inside the IDE: 

[image]

Possible reasons might be, that both mbeddr and mps-extensions are packaging the batik lib. as a stub solution. The packaging inside mps-extensions was introduced [here](https://github.com/JetBrains/MPS-extensions/pull/884). 

A possible fix would be to backport this [commit](https://github.com/mbeddr/mbeddr.core/commit/06d2162ffa536e59519146a350e200edef94bb8e) and revert batik from mbeddr.core.

arimer, 2025-04-22:
Just backporting https://github.com/mbeddr/mbeddr.core/commit/06d2162ffa536e59519146a350e200edef94bb8e will not work, because the content of the  `de.itemis.stubs.batik` from MPS-Extensions in version 2023.2 is not complete 🤦🏼 

A refactoring of the stub solution in mps-extensions is required, which was only done for MPS 2024.1.

sergej-koscejev, 2025-07-02:
@arimer is this issue now fixed?