mbeddr/mbeddr.core #2845: com.mbeddr.mpsutil.compare:AssertNodeEquals: show diff option removes output from last transient model

arimer, 2025-04-03:
MPS 2022.2
In case the show diff checkbox is set, the element under comparison is removed from the last transient model and can't be viewed any longer there. 

[image]

alexanderpann, 2025-04-03:
That sounds like such an odd issues, are you sure? The checkbox only shows a diff dialog and that's it.

arimer, 2025-04-03:
Let me show you :)

arimer, 2025-04-03:
In [addToTempModel()](http://127.0.0.1:63320/node?ref=r%3Aec874b45-e888-42e6-995a-a298cefdff55%28com.mbeddr.mpsutil.comparator.code%29%2F6212439752221681343) the node under comparison is not copied but detached from the transient model.

arimer, 2025-04-03:
>That sounds like such an odd issues, are you sure? The checkbox only shows a diff dialog and that's it.

The diff dialog is operating on a temporal model, where the node under comparison needs to be present. This praticular node should be copied and not detached.

alexanderpann, 2025-04-07:
Note in case someone wants to fix this issue: The language moved to MPS-Extensions in 2024.1.

alexanderpann, 2025-08-01:
The assert statement now also works with dynamically created nodes. The nodes are always copied to a temporary model, so this shouldn't be an issue anymore.